vsergeev/python-periphery #3: MMIO class cannot perform true 16-bit/32-bit accesses

marcan, 2016-01-06:
mmio.py uses ctyles to read/write 16- and 32-bit quantities. Unfortunately, although common sense would be that those wrappers do real 16- and 32-bit loads and stores, that's not the case. The wrappers  use memcpy, and worse, actually do read/modify/write for stores:

https://github.com/python/cpython/blob/12417ee0a6834cb874efd8061ca2d5172a0a9f42/Modules/_ctypes/cfield.c#L854

This breaks MMIO with registers that have side effects when read/written, and potentially breaks even worse on platforms where memcpy doesn't know how to coalesce writes and it ends up doing the write bytewise. mmio.py needs to find some other way to read/write registers that actually uses a single-instruction read or write.

vsergeev, 2016-02-27:
Thanks for looking into this and isolating the root cause. This is definitely a serious issue for the MMIO module. I'll think about the options that don't involve compiling and binding C code.

mpb27, 2016-09-06:
Changed the **write32()** to perform `self.mapping[offset:offset+4] = struct.pack("<L", value)` and that eliminates the read, then write issue (on my Cortex A9 platform).

I borrowed the code/idea from [https://graycat.io/tutorials/beaglebone-io-using-python-mmap/](url)

closed 2016-10-24 by vsergeev

marcan, 2016-10-24:
FWIW, you should do the same thing for reads. Though the impact is smaller, registers that are read-sensitive do exist and doing a bytewise memcpy from them will break things.

vsergeev, 2016-10-24:
@marcan I modified reads to use the same kind of access through `self.mapping[...]` in 3944894957ebc0695f166acc5cba817da6dc9500 -- is that what you mean?

marcan, 2016-10-24:
Yes, sorry, I hadn't seen that as it wasn't linked from the issue. Assuming this access method results in true 32-bit or 16-bit operations for reads/writes then all is good.

vsergeev, 2016-10-24:
@marcan OK cool, no problem. I don't have a jig to figure out what kind of instructions are being generated, but this should still be better than it was before.

mpb27, 2016-10-28:
I can confirm that at least the write32() results in one 32-bit transaction on my bus.  My application is on a ZYNQ where you can implement custom peripherals in FPGA logic.

vsergeev, 2016-10-28:
@mpb27 great, thanks for testing. Any chance you are setup to observe read32() with one of your PL peripherals?

commit dbee8a9a (2016-10-10): mmio: fix atomicity of writes

resolves #3.

thanks to Hector Martin <marcan@marcan.st> for reporting and mpb27 for
the suggested fix.
commit f24022c7 (2016-10-18): mmio: fix atomicity of writes

resolves #3.

thanks to Hector Martin <marcan@marcan.st> for reporting and mpb27 for
the suggested fix.
commit 1e3343a2 (2016-10-24): mmio: fix atomicity of writes

resolves #3.

thanks to Hector Martin <marcan@marcan.st> for reporting and mpb27 for
the suggested fix.