vsergeev/python-periphery #5: Example for using poll?

9600, 2016-05-16:
Do you have an example of using GPIO with poll? I'm trying to use this on an Intel Edison, have set edge to falling and no matter what parameter I pass to poll it always immediately returns with True, despite the state of the pin in question not having changed.

I can otherwise read and set pin states fine.

vsergeev, 2016-05-17:
There's a few examples of poll in the hardware tests here: https://github.com/vsergeev/python-periphery/blob/v1.0.0/tests/test_gpio.py#L100

Does a read followed by poll work?

jakogut, 2016-10-03:
@9600: I experienced the same issue, and I made a pull request to fix it.

vsergeev, 2016-10-03:
@jakogut thanks! I'll be updating python-periphery this week with the pending PRs and a new release.

jakogut, 2016-10-03:
@vsergeev Very cool, I'm glad to hear that such a useful library is still being maintained.

commit 0d6c24f3 (2016-10-10): gpio: make poll() more robust among platforms

resolves #7. resolves #5.
commit 022ef29a (2016-10-24): gpio: make poll() more robust among platforms

resolves #7. resolves #5.
closed 2016-10-24 by vsergeev