vsergeev/python-periphery #9: Bindings for MicroPython?

pfalcon, 2017-01-29:
Cute project! I wonder, if you would be interested one of these days to develop further hardware API bindings for MicroPython on Linux: https://github.com/micropython/micropython-lib/tree/master/machine/machine . So far, only GPIO and timer support are implemented. MicroPython has slightly different API, but it's portable to various baremetal and RTOS boards. Quick overview of the API is: http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#timers . Examples working across different types of boards: https://github.com/micropython/micropython/tree/master/examples/hwapi .

If you won't have time, I'll be glad to port your code myself some day, if you don't mind.

Thanks!

vsergeev, 2017-03-23:
Sure, I'd be happy to help. Probably a module at a time, time-permitting.

Just to clarify -- I imagine you want to micropython-lib to be dependency free, so this would be more of a straight code port, rather than wrapping python-periphery?

pfalcon, 2017-03-23:
Thanks for the reply! Yes, the idea of micropython-lib is to be lean and dependency-free. I also not sure that python-periphery as is would work with MicroPython - there certainly will be things which MicroPython misses (often on purpose, to stay small). So, there would be 2 orthogonal approaches to make python-periphery usable/useful on MicroPython:

1. Port it as is to MicroPython's language subset.
2. Reuse its known working/debugged code to implement MicroPython's native "machine" API.

Myself, I'd be interested in no.2, though if you or other python-periphery users would be interested to try no.1, I'd be interested to know about it too. Overall, this ticket is to ask if you'd have any concerns with the code reuse as described above, and see if there're any interest from more people.

Thanks!

vsergeev, 2020-03-25:
Closing due to lack of time and activity. Maybe I or someone else will be revisit this one day.

closed 2020-03-25 by vsergeev