vsergeev/python-periphery #10: GPIO pins fail to connect on first try on RaspberryPi

rwdavis513, 2017-05-05:
When initializing the pins on Raspbian, there appears to be a timing issue between the `/sys/class/gpio/export` and the write to `/sys/class/gpio/direction` 

[periphery/gpio.py](https://github.com/vsergeev/python-periphery/blob/master/periphery/gpio.py)

After running the code the first time the pin is added to the /sys/class/gpio/gpio{pin_number} list and therefore it works on the second try. 

It looks like it could be addressed by adding a wait time if it fails. Here's a wrapper which addresses the issue:

`
    for _ in range(number_of_attempts):
       try:
           gpio_pin = GPIO(pin, direction)
           break
       except GPIOError as e:
           if _+1 == number_of_attempts:
               raise e
           time.sleep(0.05)
    return gpio_pin
`

vsergeev, 2017-05-08:
Interesting, and definitely possible if the underlying driver takes some time before the GPIO pin directory is exported. Looks like there should be a `stat()`/`isdir()` loop with a timeout on the pin directory before attempting to open its `direction` file.

vsergeev, 2018-04-04:
Fixed in e6b905d.

closed 2018-04-04 by vsergeev