vsergeev/python-periphery #15: Writing to an MMIO has no effect --- what do I the wrong way?

alpintrekker, 2018-07-26:
Hi @vsergeev 

I'd power down USB0 PHY on an AM437x, so I was doing this as root from ipython console:
`from periphery import MMIO`
`USB0_CTRL_MMIO = MMIO(0x44E10620, 0x1000)`
`hex(USB0_CTRL_MMIO.read32(0x00))`
the last line gives output 
`0x3c186004`
then I executed:
`USB0_CTRL_MMIO.write32(0x00, USB0_CTRL_MMIO.read32(0x00) | 0x3)`
but reading back
`hex(USB0_CTRL_MMIO.read32(0x00))`
gives the same:
`0x3c186004`
instead of 
`0x3c186007`

Could you please help me, what do I the wrong way?

ukleinek, 2019-10-07:
Hello,

I didn't check the relevant hardware manual, but I guess the issue is just that the MMIO register you write to is readonly (or at least bits 0 and 1 are). Or maybe the register is not writable in user mode?

To rule out problems with python-periphery, you might want to use a program like [memtool](https://github.com/pengutronix/memtool). The equivalent of your program is:

    memtool md 0x44E10620
    memtool mw 0x44E10620 0x3c186007
    memtool md 0x44E10620

Best regards
Uwe

vsergeev, 2019-10-23:
I found the register on page 676 of http://www.ti.com/lit/ug/spruhl7h/spruhl7h.pdf. It looks like those bits are R/W and should retain their value. I second @ukleinek's recommendation of using a tool like `memtool` or `devmem` to write it. In addition, you could try reading and writing a scratchpad register `RTCSS_SCRATCH0_REG`.

It's also possible (but maybe not likely, in this case) that a kernel driver is overwriting the register behind the scene (e.g. if the power down caused an event that cause the driver to overwrite it).

I'm going to close this issue for now, but if you do happen to try again and see an issue with python-periphery writing data that can't be reproduced with the other tools, please re-open.

closed 2019-10-23 by vsergeev