vsergeev/python-periphery #18: Can you please explain more regarding I2C?

ishahak, 2018-10-11:
Hello,
your library looks really cool.
However I was trying to use it for I2C and could not understand the logic behind that code.
There are 3 entities involved in I2C operations: the device address, the register within that device, and the value that we want to write or read to the register.

How do you accomplish a writing of value into a register? and how do you read the value from a register?

Thank you!
-Itay

vsergeev, 2019-10-23:
Register reads and writes are implemented using a small subset of the more general I2C protocol, which consists generally of an arbitrary number of multi-byte read and/or write messages in under a single transaction to one device. python-periphery supports I2C in this sense, which allows interfacing with a wider variety of I2C devices than just the ones that implement a register map. For example, you may want to read/write multiple bytes from an EEPROM in one transaction.

It would be possible to add convenience read()/write() wrapper functions, though.

closed 2019-10-23 by vsergeev

ishahak, 2019-10-23:
Thank you very much for responding!

‫בתאריך יום ד׳, 23 באוק׳ 2019 ב-7:08 מאת ‪Vanya Sergeev‬‏ <‪
notifications@github.com‬‏>:‬