vsergeev/python-periphery #24: Question about PWM channel and pin

CharlesCCC, 2019-03-28:
Here I was trying to use this library for the Coral EdgeTPU board, I'm having such difficulty to get the PWM to work. I have post a question [here](https://stackoverflow.com/questions/55385943/google-edgetpu-cant-get-pwm-to-work-with-python). 

Here is my testing code:

    from periphery import PWM
    import time
    
    # Open PWM channel 0, pin 0
    pwm = PWM(0,0)
    
    # Set frequency to 1 kHz
    pwm.frequency = 50
    # Set duty cycle to 75%
    pwm.duty_cycle = 0.02
    
    pwm.enable()
    
    print(pwm.period)
    print(pwm.frequency)
    print(pwm.enabled)
    
    # Change duty cycle to 50%
    
    pwm.duty_cycle = 0.05
    
    pwm.close()

Problem is this part:

    # Open PWM channel 0, pin 0
    pwm = PWM(0,0)

I can see output when running `PWM(0,0)` `PWM(0,1)` `PWM(0,2)`

but I get the error messsage when trying to run the following: 

`PWM(1,1)`

`PWM(2,2)`

    mendel@elusive-jet:/sys/class/pwm$ sudo python3 /usr/lib/python3/dist-packages/edgetpuvision/testPWM.py
    OSError: [Errno 19] No such device
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.5/dist-packages/periphery/pwm.py", line 69, in _open
        f_export.write("%d\n" % pin)
    OSError: [Errno 19] No such device
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/lib/python3/dist-packages/edgetpuvision/testPWM.py", line 5, in <module>
        pwm = PWM(1,1)
      File "/usr/local/lib/python3.5/dist-packages/periphery/pwm.py", line 44, in __init__
        self._open(channel, pin)
      File "/usr/local/lib/python3.5/dist-packages/periphery/pwm.py", line 71, in _open
        raise PWMError(e.errno, "Exporting PWM pin: " + e.strerror)
    periphery.pwm.PWMError: [Errno 19] Exporting PWM pin: No such device

Based off the document from both Coral and the library site:
https://coral.withgoogle.com/tutorials/devboard-gpio/ 

https://github.com/vsergeev/python-periphery

The 

`PWM(1,1)`

`PWM(2,2)`

should have worked without issue, I can see the following directories existed:

```
"\sys\class\pwm\pwmchip0"
"\sys\class\pwm\pwmchip1"
"\sys\class\pwm\pwmchip2"
```

In the python-periphery source code 
https://github.com/vsergeev/python-periphery/blob/master/periphery/pwm.py

it should getting the path as following:

`PWM(1,1) ===> /sys/class/pwm/pwmchip1/pwm1` if `pwm1` not exists, then it should call the `export` to generate it. 

So, My main question are:

 - What is `channel` and `pin` and how it is been used ? 
 - Why I'm not able to get PWM(1,1) PWM(2,2) to work ? 

Thank you in advance.

CharlesCCC, 2019-04-02:
wrong documentation, please check #26

vsergeev, 2019-06-26:
See latest comment on #26.

commit 38926747 (2019-10-25): pwm: fix naming of channel->chip, pin->channel

resolves #24, resolves #26
commit 3b13201b (2019-10-27): pwm: fix naming of channel->chip, pin->channel

resolves #24, resolves #26
commit f261ed57 (2019-10-29): pwm: fix naming of channel->chip, pin->channel

resolves #24, resolves #26
commit 11a94677 (2019-10-29): pwm: fix naming of channel->chip, pin->channel

resolves #24, resolves #26
closed 2019-10-29 by vsergeev