vsergeev/python-periphery #32: gpio: doesn't work for named gpios

ukleinek, 2019-08-04:
Hello,

running periphery on an Arietta G25:

    root@arietta:~# python3 -c "from periphery import GPIO; GPIO(45, 'high')"
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3/dist-packages/periphery/gpio.py", line 43, in __init__
         self._open(pin, direction)
      File "/usr/lib/python3/dist-packages/periphery/gpio.py", line 82, in _open
        raise TimeoutError("Exporting GPIO: waiting for '%s' timed out" % gpio_path)
    TimeoutError: Exporting GPIO: waiting for '/sys/class/gpio/gpio45' timed out

This fails because the gpio directory appears with the name `pioB13`. That's because the driver sets names for the gpiochip in the kernel. See https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/pinctrl/pinctrl-at91.c#n1896

I don't know how this can be (easily) handled in periphery, probably the sanest approach is to use /dev/gpioctl instead of the (deprecated) /sys/class/gpio approach.

ukleinek, 2019-08-21:
FTR: My suggestion to use /dev/gpioctl is actually #13 I think. (The issue isn't very verbose, so I'm not sure.)

vsergeev, 2019-10-23:
Yeah, this has come up before with Atmel kernels on the other periphery projects, but unfortunately there is no platform agnostic way to map GPIO pin number to the exported pin directory, especially when drivers invent their own directory names in a hard-coded `kasprintf()`. See also vsergeev/c-periphery#8.

Character device GPIOs are the way to go. They're implemented in `devel` and will be released imminently in v2.0.0, along with other improvements and fixes (including your timeout=None fix).

vsergeev, 2019-10-23:
Note: it would technically be possible to pass in an extra parameter to the constructor with the expected exported pin name, but it just doesn't seem right to sully the API for a non-compliant driver.

closed 2019-10-23 by vsergeev