vsergeev/python-periphery #41: Using Python Periphery for GPIO Pins

sslmorin, 2020-06-07:
Hi! I am currently working with an UpBoard, which has multiple GPIO pins similar to a Raspberry Pi. I have three push buttons attached to three GPIO pins, and then a common ground. The circuit is really simple, but all of the elements are connected, and they read as False when the button is pushed, True otherwise. I tested this with an LED instead of my code, and it seemed to work fine. But when I run my code, even when no button is pressed it outputs as though one was. Any ideas why this could be?
`import time
import datetime
from periphery import GPIO
gpio_good_in = GPIO(26, "in")
gpio_neu_in = GPIO(6, "in") 
gpio_bad_in = GPIO(5, "in")

good_condi = gpio_good_in.read()
neu_condi = gpio_neu_in.read()  
bad_condi = gpio_bad_in.read()  
i= 1
while 1: 
if not good_condi:print(Feedback: Good)
elif not neu_condi:print(Feedback: Neutral)
elif not bad_condi:print(Feedback: Bad)
time.sleep(0.01)
i = i+1
gpio_good_in.close()  
gpio_neu_in.close()
gpio_bad_in.close()`

vsergeev, 2020-06-07:
Can you describe the connections in more detail? Do you have pull-ups on the GPIO side of the push buttons?

sslmorin, 2020-06-07:
I am very new to hardware, and had no idea pull-ups were required for this circuit. After doing a bit of reading, I can see that I will need to add them. Thank you for letting me know!

sslmorin, 2020-06-07:
I'm working on the same issue, but I noticed that there is a bias functionality built in to the GPIO class. However, when I put bias='pull_up' in the GPIO command, I get an unexpected keyword error. Would this work instead of rewiring my circuit, and if so how do I resolve the unexpected keyword error

vsergeev, 2020-06-08:
The line bias option is a relatively new feature in Linux and is only available for character device GPIOs. See https://python-periphery.readthedocs.io/en/latest/gpio.html#periphery.periphery.GPIO.CdevGPIO for the constructor.

Assuming your GPIOs are available on gpiochip0, it would look like:
``` python
gpio_good_in = GPIO(26, "/dev/gpiochip0", "in", bias="pull_up")
gpio_neu_in = GPIO(6, "/dev/gpiochip0", "in", bias="pull_up")
gpio_bad_in = GPIO(5, "/dev/gpiochip0", "in", bias="pull_up")
```

If your kernel is too old, you might get an error opening with the bias option. In that case, I would still give it a try with an external pull-up (10k - 50k should be fine) to VCC to make sure there aren't other issues.

sslmorin, 2020-06-08:
Thanks for all your help!

closed 2020-06-08 by sslmorin

vsergeev, 2020-06-09:
No problem. Feel free to reopen if you run into any other issues.