vsergeev/python-periphery #56: SPI transaction speed not meeting the maximum speed

sjpark608, 2022-08-25:
Hi,

I am using SPI module in the library to handle SPI transactions from Raspberry PI CM4 to an onboard SPI device. The maximum speed is set to 5MHz on my script, and when I performed ```spi.transfer()``` of 7 bytes, the measurement time was well above the expected time. 

I am trying to sample data at 38400SPS which is about 1 / 21us, but due to the delay of SPI transaction, I cannot actually achieve the sampling rate. I was wondering if you have any idea on how to increase the transaction speed of the SPI.

The following is my piece of code

```
spi = SPI("/dev/spidev6.1", mode = 1, max_speed = 5000000)
```
```
def read_data_bytes(spi, adc_type):
    if adc_type == 1:
        values = [RDATA1_OP] + [0xFF]*6
    else:
        values = [RDATA2_OP] + [0xFF]*6
    _logger.info(f'Sending Command = {values}')
    t1_start = perf_counter()
    values = spi.transfer(values)
    t1_stop= perf_counter()
    _logger.info(f'Reading ADC data= {values}, Tx time {t1_stop - t1_start}')
    return values
```
```
INFO:__main__:Reading ADC data= b'``\xf35{$b', Tx time 0.00021887299953959882
INFO:__main__:Data in decs = (96, 96, 243, 53, 123, 36, 98)
```

When scoped the MISO and Clock signal, I found out that the clock signal is not actually 5MHz, it's around 2MHz. And the CS is asserted much longer than it needs.

The following is captured image
[image]
[image]
[image]
[image]

sjpark608, 2022-08-26:
The clock signal being slower than the maximum clock speed set in SPI module was due to the RPI core clock divisor. I've resolved the issue by configuring dt-overlay.

I'm still having issue with the CS line being asserted and de-asserted long before and after the transmission. Is there a way to assert the line faster?

vsergeev, 2025-03-05:
Yup, `max_speed` in the userspace SPI API is just a suggestion. The SPI controller driver may end up using something lower depending on the clock configuration, as you pointed out.

As for the delay between CS assert and transfer (assuming from your code sample, you're using the SPI device's CS): it seems that the kernel driver (though I don't have your exact kernel version) uses software-controlled GPIOs for CS: https://github.com/raspberrypi/linux/blob/ac56a225e0889e60e912ecd3a51333a5aee901fc/drivers/spi/spi-bcm2835.c#L1244-L1245. So there might very well be some kernel code running between software CS assert and the hardware SPI transfer.

Closing for now, as I don't believe there is an issue here with python-periphery.

closed 2025-03-05 by vsergeev