vsergeev/python-periphery #60: strange timestamp

Skill3t, 2023-04-20:
Hey if i run the following script on Arm Cortex-A7 in a docker Container 

`from periphery import GPIO
from datetime import datetime
import time

all_gpio_list = []
c4l4 = GPIO("/dev/gpiochip4", 4, "in", edge="both")
all_gpio_list.append(c4l4)

while True:
    gpios = GPIO.poll_multiple(all_gpio_list, timeout=10)
    for gpio in gpios:
        event = gpio.read_event()
        print(event)
        print(datetime.now())
        print(time.time())
  `

the output is:
`EdgeEvent(edge='rising', timestamp=6212787543813)
2023-04-20 08:24:31.474127
1681979071.4748712`

The timestamp of the event doesn't make any sense to me and I don't know how to convert it.
According to the documentation it should be a unix timestamp in nanoseconds.
So it should be something like this datetime.fromtimestamp(event.timestamp / 1000000) unfortunately it is not.

Do you have a hint for me? 
Thanks in advance

vsergeev, 2023-04-21:
Thanks for the bug report -- it was an oversight in the new gpio-cdev v2 ABI implementation. It's been fixed in v2.4.1 now to report realtime timestamps in line events. Let me know if you run into any other issues.

closed 2023-04-21 by vsergeev

Skill3t, 2023-04-21:
Many thanks for the quick feedback. I just tested the new version. Works as expected. The solution then looks like this (for all those who also want to convert the time stamp)

datetime.fromtimestamp(event.timestamp / 1000000000)

commit 761a244c (2023-04-21): gpio: enable realtime timestamps for line events in cdev2 impl

fixes #60.